ukaea/paramak #594: Add a new reactor that includes options for ports

Shimwell, 2020-11-29:
The impact of ports on the TBR is a common request for parameter studies. Perhaps we can add another reactor based on the ball reactor that allows port to be placed

additional arguments
number of ports
width: list of floats
height: list of floats
fillet size: list of floats
port azimuthal angles: list of floats

in the event that the users does not provide the port azimuthal angles then these can be calculated by spearing them out equally. In a more realistic case they should be placed so that they line up with the gaps in the TF coils

RemDelaporteMathurin, 2020-11-30:
Should it be a new reactor or an adaptation of the existing ones ?

billingsley-john, 2020-11-30:
For ease, a new reactor based on the ball_reactor will be created first to understand the logic, then perhaps in the future the existing reactors could be adapted to include ports? However, I wouldn't be opposed to keeping completely separate reactors with and without ports, i.e. BallReactor and BallReactorWithPorts

RemDelaporteMathurin, 2020-11-30:
Yep you are right :-)

It's just that in the future it might get complex with BallReactorWithPorts, SingleNullBallReactorWithPorts, SingleNullBallReactorWithPortsWithDTFCoils, etc.. haha

But it's okay for now I guess

Shimwell, 2020-11-30:
Yep the number of reactors could really increase with all these options. This table idea can help us know what is available https://github.com/ukaea/paramak/issues/424

We might need to change the naming scheme so that all he combinations can be supported without having really long names :-)