ukaea/paramak #596: Adding fillet radius to rectangular port cutter causes problems with blanket/firstwall construction

billingsley-john, 2020-11-30:
When fillet radius of PortCutterRectangular is not none, the outboard blanket/firstwall components in a BallReactor are not constructed correctly, despite the port cutter itself being constructed correctly. Some port cuts are not performed and additional volumes are sometimes created.
[image]

RemDelaporteMathurin, 2020-11-30:
Could you provide with minimal working examples that reproduce the bug and perhaps the expected geometry ?

Shimwell, 2020-11-30:
This is a strange one. Yep would be interested in running the example as well

billingsley-john, 2020-12-01:
This is a minimal working example that breaks in quite a spectacular way.

`       test indentation`

RemDelaporteMathurin, 2020-12-01:
Thanks @billingsley-john , I'll try and have a look. This is maybe a CQ issue, in that case we should be able to isolate it.

billingsley-john, 2020-12-01:
**fillet_radius = 0**
[image]

**fillet_radius = 5
[image]
**

billingsley-john, 2020-12-01:
> Thanks @billingsley-john , I'll try and have a look. This is maybe a CQ issue, in that case we should be able to isolate it.

Just tested with cadquery 2 and it works. Must be an issue with cq master

billingsley-john, 2020-12-01:
**cadquery=2, fillet_radius = 5**
[image]

billingsley-john, 2020-12-01:
The start and stop angle of the blankets/firstwall also seems to impact if the ports are created.
start_angle = -180
stop_angle = 180
- sometimes not all ports are created (cq master AND cq 2)

start_angle = -180
stop_angle = 179
- all ports created but additional volume not cut away by center_column_cutter (cq 2)

RemDelaporteMathurin, 2020-12-01:
Does the error still occur without that center_column_cutter ?

billingsley-john, 2020-12-03:
UPDATE: @RemDelaporteMathurin error does not occur when the center_column_cutter is omitted. Also, error does not occur if the order of the cuts is switched. I.e. WORKING: ports first, then center column cutter. ERROR: center column cutter first, then ports.

billingsley-john, 2020-12-03:
WORKING EXAMPLE: Use this firstwall shape:

firstwall = p.BlanketFP(
    plasma = plasma,
    thickness = 20,
    offset_from_plasma = 50,
    start_angle = -180,
    stop_angle = 179,
    rotation_angle = 180,
    cut=[ports, center_column_cutter]
)