ukaea/paramak #628: Paramak at FOSDEM2021?

adam-urbanczyk, 2020-12-09:
Would you be interested in presenting this project at FOSDEM2021 (online): https://lists.fosdem.org/pipermail/fosdem/2020q4/003110.html . I think it would be super interesting for the community.

Shimwell, 2020-12-09:
Yes certainly I'm sure we can get something together in time.

We will have to make a presentation and clear it through the company pinboard but we have had talks on this topic cleared in the past so I don't think it will be a problem.

The FOSDEM talk you did last year was one of the reasons we investigate CQ in the first place.
https://archive.fosdem.org/2020/schedule/event/cadquery/

What does everyone else think, registration detials are here https://penta.fosdem.org/submission/FOSDEM21

RemDelaporteMathurin, 2020-12-09:
Hi all! Sounds like a great idea!

billingsley-john, 2020-12-09:
Yes, sounds great!

jmwright, 2020-12-10:
@Shimwell and team - Thanks for your willingness to do this. Fingers crossed that you get approval. Please let the CadQuery team know if you need any support from us.

Shimwell, 2021-01-14:
mini update, clearance has been received and application to FOSDEM has been submitted

jmwright, 2021-01-14:
Great! Glad to hear it.

jmwright, 2021-02-03:
@Shimwell Did you ever hear back on whether or not your submission was accepted at FOSDEM?

Shimwell, 2021-02-03:
I've just checked on the website and unfortunately it got rejected. We can make improvements and try again next year but it would be handy to know why it didn't made the cut this time.

[image]

jmwright, 2021-02-03:
It is strange that you didn't see a rejection email.

Shimwell, 2021-02-03:
No worries, it is a challenge for next year. 

More importantly I spotted a great talk that we should all go to :-)

https://fosdem.org/2021/schedule/event/cadquery/

The slides look great
https://fosdem.org/2021/schedule/event/cadquery/attachments/slides/4437/export/events/attachments/cadquery/slides/4437/FOSDEM2021_SLIDES.pdf

closing this issue for now, but I promise to reopen it in a year

closed 2021-02-03 by Shimwell

adam-urbanczyk, 2021-02-03:
You might have been too late with the submission. See the cfp https://lists.fosdem.org/pipermail/fosdem/2020q4/003110.html
```
- Call for participation closes: 20 December 2020
```

Shimwell, 2021-02-03:
That looks like it, I was certainly later than 20th of Dec. Will try to be quicker next time. Thanks Adam and looking forward to your talk