ukaea/paramak #630: New parametric component for a rotated wedge

Shimwell, 2020-12-10:
Just wondering if another parametric shapes should be added for a rotatable wedge shape.

This could be useful for cutting ports and other channels

Sketched out a few of the parameters that could be used to make the component

[image]

RemDelaporteMathurin, 2020-12-11:
I don't really get what "angle" stands for here. What's the rotation axis ?

Shimwell, 2020-12-11:
I think by default rotation axis is from the pivot point downwards on the z axis

Shimwell, 2020-12-12:
I've made a PR for this #633 it has some images the perhaps explain the shape a bit more

Shimwell, 2020-12-16:
closing as this have been merged into develop

closed 2020-12-16 by Shimwell

RemDelaporteMathurin, 2020-12-20:
@Shimwell does this belong to 0.2 instead of 0.3 ?