ukaea/paramak #631: CAD model in BREP format needed

Shimwell, 2020-12-11:
To prepare for future workflows we need a Brep file of the Cad geomentry.

This can be produced using the PPP https://github.com/ukaea/parallel-preprocessor

The future workflows this could be useful for include production of an in-printed and merged h5m file ready for using in DAGMC

The work flow would be 

export stp files -> convert to brep using PPP -> mesh with gmsh to make .msh file -> use gmsh2dagmc.py to make the h5m file

I shall get to work on producing the brep

Shimwell, 2020-12-16:
A solution was suggested on the CQ issues. I only just noticed it so I shall have a go soon.
https://github.com/CadQuery/cadquery/issues/449

billingsley-john, 2020-12-18:
Individual solids can be exported using cadquery via:

`component.solid.val().exportBrep()`