ukaea/paramak #644: Exodus support

Shimwell, 2020-12-18:
Request from potential user, could we support exodus outputs.

This could be done by including gmsh and or https://github.com/nschloe/meshio

Perhaps this should be an example instead of fully integrated into the paramak

RemDelaporteMathurin, 2020-12-20:
I think an example of how to produce meshes from CAD with GMSH and then convert it using meshio would be great. Advanced meshes need to be done with dedicated tools such as trellis, salome and so on.
meshio is really great for converting between mesh formats

GregVernon, 2021-02-04:
As a note: Trelis is now Coreform Cubit - Cubit was built around the Exodus file format and so has (I assume) the best native support for Exodus files.  We also offer a free non-commercial license, called Coreform Cubit Learn - it's fully featured, but limited to 50k elements for export.  However, as about 1/2 of Coreform are currently graduate (or undergraduate) students, we're more than happy to temporarily remove this restriction if it gets in the way of a student doing their research/thesis/dissertation.

That being said, STP export wouldn't be limited and I don't think traditional STL files would either (Cubit allows either exporting the render facets or export triangle mesh as an STL... I _think_ the former is unlimited, while the latter might still be limited).

Shimwell, 2021-02-04:
I made a script a while back that could do mesh so I guess it is just a case of changing the export statement https://github.com/ukaea/paramak/blob/main/examples/example_neutronics_simulations/make_unstructured_mesh.py

Shimwell, 2021-02-04:
ah es I can see there is a line that reads ```export abaqus "tet_mesh.inp" overwrite``` so I'm sure this script can be tidied up and implemented into the code a bit like the other faceted geometry script.