ukaea/paramak #654: Add 2D image from a cross section of the 3D geometry

Shimwell, 2021-01-01:
The current method relies on the points and doesn't show splines or circles very well.

Another method could make a large flat  slicing face (extrude it by a tiny amount) then find the intersections with each solid and save each solid as a STL file, then load the STL file and access the vertex points.

RemDelaporteMathurin, 2021-01-04:
Is this a doublon of #84 ?

Shimwell, 2021-01-25:
The issue reported in #84 is regarding the curves not being represented properly by the existing methods of export_2d_image and export_html

I think the export_html is useful for inspecting the points in by the Shape / component during development of a new component. While this slicing would be more suitable at replacing the export_2d_image method

The challenge described in this here would be a totally different way of getting a 2d image from the CAD which would also partly solve #84 but it solves it in a way that no longer involves the points.

So yes they are quite related but I shall close the other issue now as #693 solves that issue