ukaea/paramak #664: AttributeError on task 11, 1_simulate_CAD_...

steventhomas1989, 2021-01-07:
In it's current iteration, cell 3 on the notebook returns an attribute error when running the cell.

Current cell:
neutronics_model = paramak.NeutronicsModelFromReactor(
    reactor=my_reactor,
    cell_tallies=['heating'],
    source=source,
    simulation_batches=10,
    simulation_particles_per_batch=10,
    materials={
        'pf_coil_mat': 'copper',
        'outer_tf_coil_mat': 'copper',
        'blanket_mat': 'Li4SiO4'
    }
)

neutronics_model.simulate(method='pymoab')

Current error message:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-f1168e2a3824> in <module>
----> 1 neutronics_model = paramak.NeutronicsModelFromReactor(
      2     reactor=my_reactor,
      3     cell_tallies=['heating'],
      4     source=source,
      5     simulation_batches=10,

AttributeError: module 'paramak' has no attribute 'NeutronicsModelFromReactor'

Shimwell, 2021-01-07:
thanks for reporting this we shall fix it soon

Shimwell, 2021-01-07:
We have just made a new version of this task with this PR https://github.com/ukaea/openmc_workshop/pull/202

the docker image will take 1 hour to automatically rebuild but then a ```docker pull``` should have the updated task 11

Shimwell, 2021-01-27:
thanks for reporting. this is now fixed

closed 2021-01-27 by Shimwell