ukaea/paramak #665: ValueError on task 12, making_shapes_and_reactor...

steventhomas1989, 2021-01-07:
ValueError on the third cell in the notebook for task 3.

Current cell:
from paramak import RotateSplineShape

blanket = RotateSplineShape(
    points=[
        (600, 0),
        (600, -20),
        (500, -300),
        (400, -300),
        (400, 0),
        (400, 300),
        (500, 300),
        (600, 20),
        (600, 0),
    ],rotation_angle=40,
    azimuth_placement_angle=[0, 45, 90, 135, 180, 225, 270, 315]
)
blanket.solid

Current error message:
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-cbd2aafa3886> in <module>
     13         (600, 0),
     14     ],rotation_angle=40,
---> 15     azimuth_placement_angle=[0, 45, 90, 135, 180, 225, 270, 315]
     16 )
     17 blanket.solid

/opt/conda/lib/python3.7/site-packages/paramak-0.2.0-py3.7.egg/paramak/parametric_shapes/rotate_spline_shape.py in __init__(self, rotation_angle, stp_filename, stl_filename, **kwargs)
     26             stl_filename=stl_filename,
     27             connection_type="spline",
---> 28             **kwargs
     29         )

/opt/conda/lib/python3.7/site-packages/paramak-0.2.0-py3.7.egg/paramak/parametric_shapes/rotate_mixed_shape.py in __init__(self, rotation_angle, stp_filename, stl_filename, **kwargs)
     25             stp_filename=stp_filename,
     26             stl_filename=stl_filename,
---> 27             **kwargs
     28         )
     29         self.rotation_angle = rotation_angle

/opt/conda/lib/python3.7/site-packages/paramak-0.2.0-py3.7.egg/paramak/shape.py in __init__(self, points, connection_type, name, color, material_tag, stp_filename, stl_filename, azimuth_placement_angle, workplane, rotation_axis, tet_mesh, physical_groups, cut, intersect, union)
     91 
     92         self.connection_type = connection_type
---> 93         self.points = points
     94         self.stp_filename = stp_filename
     95         self.stl_filename = stl_filename

/opt/conda/lib/python3.7/site-packages/paramak-0.2.0-py3.7.egg/paramak/shape.py in points(self, values)
    458                     msg = "The coordinates of the last and first points are \
    459                         the same."
--> 460                     raise ValueError(msg)
    461 
    462                 values.append(values[0])

ValueError: The coordinates of the last and first points are                         the same.

Suggested fix:
I found that commenting out the last tuple in the list points fixed this error.

Shimwell, 2021-01-07:
thanks for reporting this we shall fix it soon

Shimwell, 2021-01-07:
We have just made a new version of this task with this PR https://github.com/ukaea/openmc_workshop/pull/202

the docker image will take 1 hour to automatically rebuild but then a ```docker pull``` should have the updated task 12

Shimwell, 2021-01-27:
thanks for reporting. this is now fixed

closed 2021-01-27 by Shimwell