ukaea/paramak #666: TypeError on Task 12, making_shapes_and_reactor...

steventhomas1989, 2021-01-07:
TypeError on the 4th cell of the current notebook.

Current cell:
from paramak import Reactor

my_reactor = Reactor()

blanket.stp_filename = 'blanket.stp'
pf_coil.stp_filename = 'pf_coil.stp'
center_column.stp_filename = 'center_column.stp'

blanket.material_tag = 'blanket_material'
pf_coil.material_tag = 'pf_coil_material'
center_column.material_tag = 'center_column_material'

my_reactor.add_shape(blanket)
my_reactor.add_shape(pf_coil)
my_reactor.add_shape(center_column)

Current error message:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-5-e4978b48cdab> in <module>
      1 from paramak import Reactor
      2 
----> 3 my_reactor = Reactor()
      4 
      5 blanket.stp_filename = 'blanket.stp'

TypeError: __init__() missing 1 required positional argument: 'shapes_and_components'

Suggested fix:
from paramak import Reactor

blanket.stp_filename = 'blanket.stp'
pf_coil.stp_filename = 'pf_coil.stp'
center_column.stp_filename = 'center_column.stp'

blanket.material_tag = 'blanket_material'
pf_coil.material_tag = 'pf_coil_material'
center_column.material_tag = 'center_column_material'

my_reactor = Reactor([blanket, pf_coil, center_column])

If you initialise Reactor with an empty list, it later states that "AttributeError: 'Reactor' object has no attribute 'add_shape'" so my suggested fix takes this into account.

Shimwell, 2021-01-07:
We have just made a new version of this task with this PR https://github.com/ukaea/openmc_workshop/pull/202

the docker image will take 1 hour to automatically rebuild but then a ```docker pull``` should have the updated task 12

Shimwell, 2021-01-27:
thanks for reporting. this is now fixed

closed 2021-01-27 by Shimwell