ukaea/paramak #667: ImportError on Task 12, making_shapes_and_reactor...

steventhomas1989, 2021-01-07:
ImportError on cell 10 of the current version of the notebook.

Current cell:
from parametric_plasma_source import Plasma

my_plasma = Plasma(elongation=2.9,
                   minor_radius=1.118*100,
                   major_radius=1.9*100,
                   triangularity = 0.55)
my_plasma.export_plasma_source('my_custom_plasma_source.so')

Current error message:
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-10-6fc27b66a9b6> in <module>
----> 1 from parametric_plasma_source import Plasma
      2 
      3 my_plasma = Plasma(elongation=2.9,
      4                    minor_radius=1.118*100,
      5                    major_radius=1.9*100,

ImportError: cannot import name 'Plasma' from 'parametric_plasma_source' (/opt/conda/lib/python3.7/site-packages/parametric_plasma_source/__init__.py)

Shimwell, 2021-01-07:
We have just made a new version of this task with this PR https://github.com/ukaea/openmc_workshop/pull/202

the docker image will take 1 hour to automatically rebuild but then a ```docker pull``` should have the updated task 12

Shimwell, 2021-01-27:
thanks for reporting. this is now fixed

closed 2021-01-27 by Shimwell