ukaea/paramak #672: Add the ability to convert existing stp files into Shapes

Shimwell, 2021-01-10:


Shimwell, 2021-02-20:
This is now built into the utils which makes things a little simpler

```
import paramak

solid, wires = paramak.utils.load_stp_file('original_stp_files/FirstWall.stp', scale_factor=0.1)

```

But I think a nicer option would be a ```ShapeFromFile``` class

RemDelaporteMathurin, 2021-03-03:
@Shimwell if we'd want to cut a shape with this imported solid would something like this work ?
ping @jhdark
```python
import paramak

# import the cad file
solid, wires = paramak.utils.load_stp_file('my_file.stp', scale_factor=0.1)
my_imported_shape = paramak.Shape()
my_imported_shape.solid = solid

# create a blanket and cut it with the imported shape
blanket = paramak.BlanketFP(......)
blanket.cut = my_imported_shape
```

Shimwell, 2021-03-03:
I think that will work and if it doesn't then we can add some code to the cut logic. Sometimes we need code to handel solids a little differently (cad query sometimes needs .val or not .val for different types of solids)