ukaea/paramak #679: testing importError with new CI

Shimwell, 2021-01-14:
Testing of import errors would ideally be implemented into the test suite. 

We don't currently test the following lines

```
    except (ImportError, ModuleNotFoundError):
        msg = "Conversion to VTK requested," \
            "but the Python VTK module is not installed."
        raise ImportError(msg)
```

In this particular case this vtk package is needed for the neutronics but not needed for geometry construction.

So we could have another CI that performs geometry only tests and check these.

The new CI could be based on a Docker image that does not include any neutronics dependancies

RemDelaporteMathurin, 2021-01-14:
That could be an idea. But then again i'm not convinced it is worth it ? it would mean a whole new test suite for just for checking that an ImportError is raised when trying to import a missing module. 
Plus it is worth noting that an ImportError is already raised in python when trying to import a missing module so doing 
```python
try:
   import module
except (ImportError, ModuleNotFoundError):
   raise ImportError
```

is a bit redundant to me.

Additionnaly, users using the Docker environnement won't ever (in theory) experience this issue. Only pip users but that's untestable atm (plus I don't see vtk in requirements.txt or setup.py).

For these reasons, I would say we don't have these kinds of try/except for imports but just imports at the beginning of the function. 👼

Shimwell, 2021-01-14:
interesting, what do you think about making the import of neutronics utils and NeutronicsModel within the __init__.py depend upon the existence of these extra packages

RemDelaporteMathurin, 2021-01-15:
Importing them at all times and only importing the packages when the functions are called is really fine.

Just not sure of the added value of the try/except thing. I think having a simple `import module` is sufficient: we don't bother with additionnal testing suite just to test this and we have our coverage up 💚