ukaea/paramak #680: Some assertions not being tested in some unit tests

billingsley-john, 2021-01-15:
Some self.assertRaises tests are not being performed when they are bundled together with others in the same unit test.

For example:

`def test_invalid_parameters_errors(self):`
`        def incorrect_inner_radius():`
`        def incorrect_mid_radius():`

`        self.assertRaises(ValueError, incorrect_inner_radius)`
`        self.assertRaises(ValueError, incorrect_mid_radius)`

In the example above, only `incorrect_inner_radius` will actually be tested. `incorrect_mid_radius` is not performed.

billingsley-john, 2021-01-15:
UPDATE: The issue raised is partially correct. Indeed, we get false-positive assertions but this is due to the first 'sub-test', i.e. incorrect_inner_radius(), running into the second 'sub-test', i.e. incorrect_mid_radius(). If the changes in the first sub-test are designed to raise ValueError, the second sub-test will always raise ValueError also due to the changes in the first. I.e. the ValueError raised in the second test is not necessarily due to the parameter we want to test.

Shimwell, 2021-01-15:
Interesting find here @billingsley-john wondering which file this is

billingsley-john, 2021-01-15:
Resolved in #659

closed 2021-01-15 by Shimwell