ukaea/paramak #712: Getting Cubit  / Trelis into the CI

Shimwell, 2021-02-04:
Currently we test about 99% of the code in the CI on github actions and circle CI but we are not able to test the Trelis scripts for making the faceted geometry.

This is because as far as I know there is no version (including the new free version) that can be downloaded from a URL without login to the website. Any suggestions appreciated here @GregVernon 

Ideally the would be a public URL for Trelis download then we could incorporate it into the Dockerfiles that are used for CI

We have a dockerfile that makes use of locally downloaded installer files (.deb) but naturally we can't put these on the internet as that would be redistribution.

But here is the dockerfile that adds to the exisintg CI dockerfile for interest

```docker
FROM ukaea/paramak

# this section copies over the dagmc plugin for trelis, a public url can be made for this open source plugin
COPY trelis17.1_ubuntu18.04.tar.gz .
RUN mkdir -p /opt/Trelis-17.1/
RUN tar -xf trelis17.1_ubuntu18.04.tar.gz -C /opt/Trelis-17.1

#this command copies the locally downloaded trelis into the docker image but a wget command pointing to a public url would allow us to do this in the CI
COPY Trelis-17.1.0-Lin64.deb .
RUN dpkg -i Trelis-17.1.0-Lin64.deb

# this installs the dagmc plugin
RUN cd /opt/Trelis-17.1/bin && \
    ln -s plugins/svalinn/libdagmc.so . && \
    ln -s plugins/svalinn/libiGeom.so . && \
    ln -s plugins/svalinn/libmakeWatertight.so . && \
    ln -s plugins/svalinn/libmcnp2cad.so . && \
    ln -s plugins/svalinn/libMOAB.so . && \
    ln -s plugins/svalinn/libMOAB.so.5 .

# puts the path for trelis into the PATH env
ENV PATH=$PATH:/opt/Trelis-17.1/bin
```

GregVernon, 2021-02-04:
There will very soon be Coreform Cubit downloads available from a public URL.  We are _actively_ working on this in our current sprint and hope to have this available at the end of this sprint (~10 more days) or the next.

Shimwell, 2021-02-16:
@GregVernon how is that sprint going, wondering if the public URL make it to the deliverables this round

GregVernon, 2021-02-17:
Hi @Shimwell - the sprint went well!  We're _very_ close to having this, in fact, we could probably push this live if we wanted, but we're going to take another sprint to stress-test and run QA on the process.

Here's a screenshot of our staged website for downloads:
[image]

Shimwell, 2021-02-17:
This is the most exciting news. Thanks for the update, the staged website looks perfect.

Shimwell, 2021-02-18:
Just to add there is some really nice work on the DAGMC repository where GitHub actions is being put to great use. 

Automatically building the DAGMC plugin on multiple operating systems and versions of Trelis Cubit

The last successful GitHub action has a artifact that is the DAGMC plugin for Cubit 2020.2 
https://github.com/svalinn/Trelis-plugin/actions/runs/576252484

Open URL for the Coreform Cubit could simplify this GitHub action and remove the need for various GitHub secrets that are used at the moment. The other thing that would simplify this plugin building is the SDK which is also a bit tricky to download on CI at the moment.

Shimwell, 2021-03-18:
Looks like cubit learn linux makes use of cent OS 7.

Currently we use Ubuntu so would prefer a .deb package

GregVernon, 2021-04-01:
@Shimwell - our latest dev-builds are now coming off the pipelines with .deb packages.  We're not perfect there yet, but we're making progress :)

Shimwell, 2021-04-03:
Thank Greg, I shall give them a go and thanks for letting me know. ps looking forward to your talk on Cubit Core Form and Python

Shimwell, 2021-04-03:
Just tried installing cubit deb package on Ubutunt 18.04 and got a seg fault so I shall try Ubuntu 20.04 and also some of the suggestions from the corefore forum [image]

But super to see progress is being made