ukaea/paramak #739: reactor workplane cannot be changed

billingsley-john, 2021-02-23:
Currently, reactor workplane cannot be specified. It is defaulted to XY - should this be controllable?

Shimwell, 2021-02-23:
Any chance you could provide an example as I didn't even think reactors had workplaces.

billingsley-john, 2021-02-23:
Apologies - this wasn't described very well. Currently, reactors do not have a workplane option so their orientation is dictated by the default workplanes of all shapes and components that make up the reactor. Currently, export_svg defaults to a workplane which views reactors from the top-down, but the reactor workplane can't be changed. However, instead of adding workplane control to the reactors, perhaps this issue is better resolved in the export_svg function.

closed 2021-02-23 by billingsley-john