ukaea/paramak #741: Simplify the shape.py points setter

Shimwell, 2021-02-23:
Currently the setter in the shape.py checks quite a lot of things about the points

https://github.com/ukaea/paramak/blob/785c3ed7304e22eac7d58bb1bdc6515fbd20b9a8/paramak/shape.py#L412-L471

This should be simplified and some of the logica can be moved into parametric shape  classes that inherit from shape.

For example this could be added to the RotateStraightShape.py as a new setter that makes use of the existing points setter in Shape.py but also extends it

```
    @RotateMixedShape.points.setter
    def points(self, values):
        if values is not None:
            # Checks that the length of each tuple in points is 2
            for value in values:
                if len(value) != 2:
                    msg = "individual points must contain 2 coordinates {} has a \
                        length of {}".format(value, len(values[0]))
                    raise ValueError(msg) 

        super(RotateStraightShape, self.__class__).points.fset(self, values)
```

RemDelaporteMathurin, 2021-05-10:
@Shimwell just to be clear, in the snipset you posted, shouldn't the test for MixedShape points ensure values is a list of list of length 3 (X, Y, connection)?

Shimwell, 2021-05-10:
> @Shimwell just to be clear, in the snipset you posted, shouldn't the test for MixedShape points ensure values is a list of list of length 3 (X, Y, connection)?

ah yes looks like I got the names mixed up. The general idea is that we could move some checking to other classes which add extra checks to the check in the Shape.py class. These extra checks would be run and then the basic checks from Shape.py are also run via the super call

Shimwell, 2021-05-10:
I am not sure if this is worth the effort as I think in general it would increase the lines of code but perhaps simplify the code a bit

RemDelaporteMathurin, 2021-05-10:
Yes some of the checks apply to all parametric shapes (hence why they were in Shape in the first place)