ukaea/paramak #747: boolean operation ordering

billingsley-john, 2021-02-26:
Currently, the order in which boolean operations are performed is hard coded in the reactor.py class:
- Cut
- Intersect
- Union
However, sometimes these operations are not mutable for the desired shape.
Control of the order in which these operations are performed could be useful.

Shimwell, 2021-02-26:
Nice suggestion, I think Remi is also keen on this. I guess we could combine the operations into a dictionary and cycle through them in the user specified order.

RemDelaporteMathurin, 2021-02-27:
I needed that a while ago. I found a workaround by rewriting the create_solid method for a shape but being able to control  the operations order would be very handy!

Shimwell, 2021-02-27:
Sounds like a popular feature request. Lets add it to the project plan