ukaea/paramak #750: new parametric shape

Shimwell, 2021-02-27:
I keen having to make this extruded slice shape and was thinking it would be handy to have a new parametric shape to make this easier

Currently I am doing the following

```
sector_angle_size = 20

cutting_wedge = paramak.CuttingWedge(
            height=2000,
            radius=2000,
            rotation_angle=360-sector_angle_size,
            azimuth_placement_angle=angle-(0.5*sector_angle_size)
    )

paramak.ExtrudeMixedShape(
    points = ....
    azimuth_placement_angle= angle-sector_angle_size
    cut = cutting_wedge,
    distance= 40
)

```

But it would be great to have something like this

paramak.ExtrudedSlice(
    angle=20,
    azimuth_placement_angle = ...
    distance= 40
)

[image]

RemDelaporteMathurin, 2021-02-27:
@Shimwell I don't really get what is extruded here ?

Shimwell, 2021-02-27:
Perhap I am getting mixed up here but we can make a slice of cake easily. The cake has a rounded rotated edge. This one has a straight edge instead.

The innertfcoilsflat is similar but only allows a simple cross section. I am keen on mixed connections
https://paramak.readthedocs.io/en/main/paramak.parametric_components.html#innertfcoilsflat

[image]