ukaea/paramak #751: points are not reusable between Shapes

Shimwell, 2021-02-27:
Points is passed by the user and changed internally by the class (perhaps this is poor program design).

One change adds the start point as the end point to make a closed wire. This is nice as it saves the user time and having to make that last point which the program can easily figure out.

It also checks that the input first and last points are not the same.

Anyway the points is pass as a list which is mutable.

Therefore the list values are changed and can't be reused. This simple example produces an error.

Perhaps we should not change the points that the user inputs by make a copy and change the copy?

```python
points =[
        (100, 0, "straight"),
        (200, 0, "straight"),
        (250, 50, "straight"),
        (200, 100, "straight"),
    ]

test_rotate_mixed_shape = paramak.RotateMixedShape(
    rotation_angle=180,
    points=points
)
test_extrude_mixed_shape = paramak.ExtrudeMixedShape(
    distance=10,
    points=points
)
```

```python
ValueError                                Traceback (most recent call last)
<ipython-input-9-eeb600e00657> in <module>
     10     points=points
     11 )
---> 12 test_extrude_mixed_shape = paramak.ExtrudeMixedShape(
     13     distance=10,
     14     points=points

~/paramak/paramak/parametric_shapes/extruded_mixed_shape.py in __init__(self, distance, extrude_both, rotation_angle, extrusion_start_offset, stp_filename, stl_filename, **kwargs)
     33     ):
     34 
---> 35         super().__init__(
     36             stp_filename=stp_filename,
     37             stl_filename=stl_filename,

~/paramak/paramak/shape.py in __init__(self, points, connection_type, name, color, material_tag, stp_filename, stl_filename, azimuth_placement_angle, workplane, rotation_axis, tet_mesh, surface_reflectivity, physical_groups, cut, intersect, union)
     94 
     95         self.connection_type = connection_type
---> 96         self.points = points
     97         self.stp_filename = stp_filename
     98         self.stl_filename = stl_filename

~/paramak/paramak/shape.py in points(self, values)
    465                     msg = "The coordinates of the last and first points are \
    466                         the same."
--> 467                     raise ValueError(msg)
    468 
    469                 values.append(values[0])

ValueError: The coordinates of the last and first points are                         the same.
```

RemDelaporteMathurin, 2021-02-27:
I noticed that a while ago, and it was a tad annoying 😸

Shimwell, 2021-05-10:
Another related problem is that if one wants to reuse the points in another shape then I think they will have been modified by the first shape and potentially unusable in the second shape

```python
shape1 = paramak.RotateStraightShape(
    points= [
    (100, 0),
    (200, 0),
    (250, 50),
    (200, 100),
]
)

# shape1.points is now a list with connections not what the user originally input

shape1 = paramak.RotateSplineShape(
    points= shape1.points
]
)
```

RemDelaporteMathurin, 2021-05-10:
One way to solve this would be to have two separate attributes `points` and `connections`.

One would only contain coordinates and the latter will only contain strings?

Shimwell, 2021-05-10:
Or perhaps another version of points called processed_points or something like that

RemDelaporteMathurin, 2021-05-28:
@Shimwell I think this can be closed since it was solved in #839

PR #839 (2021-05-10): Reuse points between shapes
## Proposed changes

This fixes #751 

## Types of changes

What types of changes does your code introduce to the Paramak?
_Put an `x` in the boxes that apply_

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code refactoring
- [ ] Documentation Update (if none of the other choices apply)
- [ ] New tests

## Checklist

_Put an `x` in the boxes that apply. You can also fill these out afte [... cut]