ukaea/paramak #753: Speed up graveyard creation

Shimwell, 2021-03-01:
The graveyard creation can be slow as it loops through every shape in the Reactor

Perhaps adding an optional argument to the reactor.make_graveyard method that allows a subset of shapes to be searched would be helpful.

This should cut down on the number of components that the bounding box is constructed for and therefore reduce the compute time required.

Might be best to either do this after #743 is merged on a branch from then #743 branch as there are some changes to the Graveyard filename in that PR

Shimwell, 2021-03-01:
This was found in #733 which is failing CI because the graveyard creation is slow

RemDelaporteMathurin, 2021-03-05:
> This was found in #733 which is failing CI because the graveyard creation is slow

In the end it wasn't because of that, still this is a great improvement.