ukaea/paramak #755: Notebook container not working

RemDelaporteMathurin, 2021-03-03:
Step to reproduce
Run the following command
`docker run -p 8888:8888 ukaea/paramak /bin/bash -c "jupyter notebook --notebook-dir=/opt/notebooks --ip='*' --port=8888 --no-browser" --allow-root`

Error message:
```
Unable to find image 'ukaea/paramak:latest' locally
latest: Pulling from ukaea/paramak
852e50cd189d: Already exists
a6236801494d: Already exists
679c171d6942: Already exists
1905567ded66: Pull complete
dc7eaf90d0c3: Pull complete
8cc47e9db86e: Pull complete
ad462465031a: Pull complete
695c60947ecd: Pull complete
4d04dc2200f5: Pull complete
d1b2873e4711: Pull complete
d1018ebba1d3: Pull complete
092b318373bc: Pull complete
5c15816cfa19: Pull complete
8a40e7d81fdf: Pull complete
56d500db08d2: Pull complete
9fce1e9a2ca3: Pull complete
aeda7b44c65b: Pull complete
3e06ef393692: Pull complete
2e2857408188: Pull complete
2676e1dd7dee: Pull complete
44055dd56d57: Pull complete
06dc4e669d34: Pull complete
68b4148f7f27: Pull complete
3b0c8058d21c: Pull complete
Digest: sha256:b2a6777f74c57df126e914a5bdabc3890bb449bebcd46de7d8e3c872d844c7c3
Status: Downloaded newer image for ukaea/paramak:latest
[C 08:55:28.878 NotebookApp] Bad config encountered during initialization: No such notebook dir: ''/opt/notebooks''
```

Shimwell, 2021-03-03:
perhaps duplicate of #749

Shimwell, 2021-03-03:
Perhaps try without the ```--notebook-dir=/opt/notebooks``` part

RemDelaporteMathurin, 2021-03-03:
This is the new command I ran:
`docker run -p 8888:8888 ukaea/paramak /bin/bash -c "jupyter notebook --ip='*' --port=8888 --no-browser" --allow-root`

and the new message error
```
[I 09:15:35.399 NotebookApp] Writing notebook server cookie secret to /root/.local/share/jupyter/runtime/notebook_cookie_secret
[W 09:15:35.575 NotebookApp] WARNING: The notebook server is listening on all IP addresses and not using encryption. This is not recommended.
[C 09:15:35.578 NotebookApp] Running as root is not recommended. Use --allow-root to bypass.
```

The `--allow-root` flag must be placed in quotes. Running: 
`docker run -p 8888:8888 ukaea/paramak /bin/bash -c "jupyter notebook --ip='*' --port=8888 --no-browser --allow-root"`

works.

However, the displaying of the solid attribute doesn't seem to work.

[image]

Shimwell, 2021-03-03:
oh yes the solid attribute needs a bit of work, I think something changed in the CQ2.1

We should look into adding this to the docker image https://github.com/bernhard-42/jupyter-cadquery

Shimwell, 2021-03-03:
this is being worked on in #756

RemDelaporteMathurin, 2021-05-10:
@Shimwell maybe we can close this?