ukaea/paramak #757: Can't cut SplineShape with another SplineShape when azimuth_placement_angle=0

RemDelaporteMathurin, 2021-03-04:
When trying to cut an ExtrudeSpline

**How to reproduce**

```python
import paramak 
shape1 = paramak.RotateSplineShape(
    points=[
        (1.1 + 0, 0),
        (1.1 + 0.5, 0),
        (1.1 + 0.5, 0.5),
        (1.1 + 0, 0.5),
    ],
    rotation_angle=360
)

shape2 = paramak.ExtrudeSplineShape(
    points=[
        (0 + 1, 0),
        (1 + 1, 0),
        (1 + 1, 1),
        (0 + 1, 1),
    ],
    distance=0.5,
    azimuth_placement_angle=0,
    cut=shape1
)

shape2.solid
```

These alterations work:

**Using StraightShape**
```python
import paramak

shape1 = paramak.RotateStraightShape(
    points=[
        (1.1 + 0, 0),
        (1.1 + 0.5, 0),
        (1.1 + 0.5, 0.5),
        (1.1 + 0, 0.5),
    ],
    rotation_angle=360
)

shape2 = paramak.ExtrudeSplineShape(
    points=[
        (0 + 1, 0),
        (1 + 1, 0),
        (1 + 1, 1),
        (0 + 1, 1),
    ],
    distance=0.5,
    azimuth_placement_angle=0,
    cut=shape1
)

shape2.solid
```
or
```python
import paramak

shape1 = paramak.RotateSplineShape(
    points=[
        (1.1 + 0, 0),
        (1.1 + 0.5, 0),
        (1.1 + 0.5, 0.5),
        (1.1 + 0, 0.5),
    ],
    rotation_angle=360
)

shape2 = paramak.ExtrudeStraightShape(
    points=[
        (0 + 1, 0),
        (1 + 1, 0),
        (1 + 1, 1),
        (0 + 1, 1),
    ],
    distance=0.5,
    azimuth_placement_angle=0,
    cut=shape1
)

shape2.solid
```
**Modifying azimuth_placement_angle**

```python
import paramak 
shape1 = paramak.RotateSplineShape(
    points=[
        (1.1 + 0, 0),
        (1.1 + 0.5, 0),
        (1.1 + 0.5, 0.5),
        (1.1 + 0, 0.5),
    ],
    rotation_angle=360
)

shape2 = paramak.ExtrudeSplineShape(
    points=[
        (0 + 1, 0),
        (1 + 1, 0),
        (1 + 1, 1),
        (0 + 1, 1),
    ],
    distance=0.5,
    azimuth_placement_angle=90,
    cut=shape1
)

shape2.solid
```

**Modifying rotation angle**

```python
shape1 = paramak.RotateSplineShape(
    points=[
        (1.1 + 0, 0),
        (1.1 + 0.5, 0),
        (1.1 + 0.5, 0.5),
        (1.1 + 0, 0.5),
    ],
    rotation_angle=180
)

shape2 = paramak.ExtrudeSplineShape(
    points=[
        (0 + 1, 0),
        (1 + 1, 0),
        (1 + 1, 1),
        (0 + 1, 1),
    ],
    distance=0.5,
    azimuth_placement_angle=0,
    cut=shape1
)

shape2.solid
```

This issue is causing #733 tests to fail.

@Shimwell @billingsley-john have you ever experienced that kind of issue ?

Shimwell, 2021-03-04:
I wonder if this is similar to the port cutting problem John observed

billingsley-john, 2021-03-04:
I have managed to produce the shape locally.

[image]

RemDelaporteMathurin, 2021-03-04:
> I have managed to produce the shape locally.
> 
I'm using the latest ukaea/paramak:dependencies image. Shall I try and run this in the CI (open a PR ?)

billingsley-john, 2021-03-04:
> > I have managed to produce the shape locally.
> 
> I'm using the latest ukaea/paramak:dependencies image. Shall I try and run this in the CI (open a PR ?)

Yes, could be a quick way to check. I'm not sure which version of the paramak and cadquery that image is using. I think i'm using a fairly recent version of the cadquery main branch

RemDelaporteMathurin, 2021-03-04:
I'll do that then. My image was pulled yesterday

billingsley-john, 2021-03-04:
Just realised the CI uses that docker image too, so it will most likely fail in the CI as well. I will pull the image and have a look

RemDelaporteMathurin, 2021-03-04:
I've opened a PR #758

billingsley-john, 2021-03-04:
> I've opened a PR #758

Looks like it's failing. I got it to work with CQ2.1RC1 which is an older version of CQ2.1. Will test out some different versions

billingsley-john, 2021-03-04:
Matrix testing for different cadquery versions is being added to actions CI #760. This should help solve these types of problems