ukaea/paramak #783: No material dt_plasma found

AnderGray, 2021-03-22:
I'm trying to run a parametric neutronics example with OpenMc & DAGMC.

I think I've done the installation correctly, but when trying to run the examples "ball_reactor.py" and "ball_reactor_minimal.py" I get the following error:

Implicit Complement assumed to be Vacuum
 ERROR: No material dt_plasma found for volume (cell) 1

Any idea what is going on?

Cheers,
A

Shimwell, 2021-03-22:
Hi Ander

Is this from the main branch, if so then I think it might require an an additional line in the NeutronicsModel materials

I think we remove the plasma from the simulation on the develop branch
```
    neutronics_model = paramak.NeutronicsModel(
        geometry=my_reactor,
        source=source,
        materials={
            'inboard_tf_coils_mat': 'copper',
            'center_column_shield_mat': 'WC',
            'divertor_mat': 'eurofer',
            'firstwall_mat': 'eurofer',
            'blanket_mat': blanket_material,  # use of homogenised material
            'blanket_rear_wall_mat': 'eurofer',
            'DT_plasma': 'dt_plasma'
        },
        cell_tallies=['TBR'],
        simulation_batches=5,
        simulation_particles_per_batch=1e4,
    )
```

AnderGray, 2021-03-22:
Hi Jon,

thanks for the quick response. It works with your edit.

For ball_reactor_minimal.py I'm getting:

Leakage Fraction            = 0.56627 +/- 0.00325

Cheers!

closed 2021-03-22 by AnderGray