ukaea/paramak #784: Example for varible offset first wall

Shimwell, 2021-03-24:
[image]

Threw together a quick example for a variable offset firstwall that might be handy to add to the examples scripts as a PR

```python

from numpy.lib.function_base import angle
import paramak
import random
import subprocess

subprocess.call(['rm', '*.svg'])

for counter in range(10):
    offset_1 = random.uniform(10, 80)
    offset_2 = random.uniform(10, 80)
    offset_3 = random.uniform(10, 80)
    offset_4 = random.uniform(10, 80)
    offset_5 = random.uniform(10, 80)

    offests = [offset_1, offset_2, offset_3, offset_4, offset_5, offset_1]

    firstwall = paramak.BlanketFP(
        rotation_angle=35,
        thickness=1,
        start_angle=0,
        stop_angle=360,
        offset_from_plasma=offests
    )
    filename = 'random_offset_'+str(counter).zfill(3)+'.svg'
    firstwall.export_svg(
        filename=filename,
        projectionDir=(5,-5,0),
        showHidden=True,
    )

subprocess.check_call(
    ["convert", "-delay", "40", "random_offset_*.svg", "random_offsets.gif"])
111

RemDelaporteMathurin, 2021-03-25:
It seems the thickness of the blanket is not constant. Is it normal ?

billingsley-john, 2021-03-25:
> It seems the thickness of the blanket is not constant. Is it normal ?

I think it looks like the thickness is not constant because of the viewing plane of the svg image. Exporting stp files and viewing on correct plane shows constant thickness

RemDelaporteMathurin, 2021-05-28:
@billingsley-john @Shimwell I get it, this is a compound of a plasma and a very very thin blanket.

Maybe runing it with a thickness of 10 makes the distinction between plasma and blanket clearer?

[image]