ukaea/paramak #791: odd behaviour with port cutter  center_point

py1sl, 2021-03-29:
I may just be mis-understanding the intended behaviour

Actual behaviour : 
When changing the first part of the center_point variable (**A**, b)  the ports radiate out from a central point at a different height.  When changing the 2nd  part of the center_point variable (A, **b** )  the resulting solids do not look as expected, they do not all radiate from the same central point.    Looks to effect both circular and rectangular port cutters

Expected behaviour: 3 solids radiating from a  single point
branch: develop
```

component_PCC = paramak.PortCutterCircular(
        distance=10,
        center_point=(0, 10),
        radius=0.1,
        azimuth_placement_angle=[0, 45, 90],
    )

```

billingsley-john, 2021-03-30:
Hi, thank you for raising this issue - perhaps the name of this variable should be changed. When (**A**, b) is specified, this translates the port in the "X" direction (depending on workplane). I.e. center_point specifies the center point of the rectangular/circular face/cross-section of the port. 
As azimuthal_placement_angle is relative to the origin, the ports will not radiate from the same central point if they have been translated.
These diagrams may be useful for illustrating this.

center_point = (0, 0)

[image]

center_point = (A, B)

[image]

py1sl, 2021-03-30:
Many thanks for the quick response and helpful diagrams, it appears i had misunderstood the intended behaviour.

closed 2021-03-30 by py1sl