ukaea/paramak #818: reactor.export_stp() to allow single stp file creation

Shimwell, 2021-04-23:
Currently the reactor.export_stp() will export separate stp files for each shape. This is done for preserving the material identity of each volume. However not every user cares about the material identity so it would be useful if the stp export can export a single stp file containing all the volumes / shapes. 

Perhaps this should even be the default behavior?

```python
reactor.export_stp(single_file=True)
```

RemDelaporteMathurin, 2021-04-26:
That is a brilliant idea. perhaps instead of "single_file" we could have something like "merge"?

Shimwell, 2021-04-26:
Perhaps we can just use the arg ```filename```. If a filename is provided then export to one file else export to separate files each named with the shape.stp_filename attrobutes

RemDelaporteMathurin, 2021-04-26:
Even better.
Can one sets a folder but still have seperate files?

Shimwell, 2021-04-26:
yes, output_folder can be set and the separate stp files will be saved there. 

output_folder can be ignored if filename is set as the filename can contain the full path

Shimwell, 2021-05-17:
solve in #819

closed 2021-05-17 by Shimwell