ukaea/paramak #825: cannot use same custom defined material for different components

billingsley-john, 2021-04-29:
I have been using the NMM to assign custom materials to reactor components. I have noticed that the same custom material cannot be assigned to several different material tags.

E.g.
my_material = nmm.Material()

my_model = p.NeutronicsModel(
        materials = {
                "material_tag_1": my_material,
                "material_tag_2": my_material
        }
)

Only material_tag_1 is entered into the materials.xml file, resulting in a simulation error as the material for material_tag_2 cannot be found.

billingsley-john, 2021-04-29:
(On develop branch)

Shimwell, 2021-04-29:
Ah yes sorry about this,

The underlying reason is because the tallies are linked to a specific material and the materials are used to identify each shape.

https://github.com/ukaea/paramak/blob/d477a7a048a7f3f209e2ba4a5ba0b815700ffbc6/paramak/parametric_neutronics/neutronics_model.py#L584

I hope to move it to volume based identification in the future but this is a core change to the assignment and reading of tallies.

Currently the work around is to make another material which is the same but a different object

Shimwell, 2021-04-29:
This is related to https://github.com/ukaea/paramak/issues/816

closed 2021-04-29 by billingsley-john